pietern/goestools #151: fatal error: proj_api.h: No such file or directory

ZL1LAC, 2022-10-17:
Not sure what am doing wrong?  It always worked.  

OS: Ubuntu 22.04.1 LTS x86_64
Kernel: 5.15.0-50-generic

```
[ 89%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/handler_goesn.cc.o
In file included from /home/server/goestools/src/goesproc/map_drawer.h:5,
                 from /home/server/goestools/src/goesproc/handler_goesn.cc:12:
/home/server/goestools/src/goesproc/proj.h:16:10: fatal error: proj_api.h: No such file or directory
   16 | #include <proj_api.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/goesproc/CMakeFiles/goesproc.dir/build.make:174: src/goesproc/CMakeFiles/goesproc.dir/handler_goesn.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2338: src/goesproc/CMakeFiles/goesproc.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

gvfd107, 2022-10-18:
1 of two things. Installing python broke mine because it put libproj in
another directory.

2 ubuntu 22 also broke libproj for me so i had to dial back to atleast 20
for a fresh install.

ZL1LAC, 2022-10-18:
Going back to Ubuntu 18.04.6 LTS fixed it for me.

closed 2022-10-18 by ZL1LAC

sarbo, 2022-10-19:
Check   the  GitHub    repo  ……     I  believe there is  a  Docker  file  that  works for me  on Mac.    I have  not  run GOES in  a while   but I  can’t build  it on mac  .  I use  the  Docker  container to  run it.

Stephen