pietern/goestools #155: Illegal instruction

wwoofbum, 2022-12-15:
Compiled & run successfully on Raspberry Pi 4B, under PiOS Buster (after failing to compile under Bullseye).  Moving SD card to a Pi 3B, all else the same, executing 'goesrecv -v -i 1 -c ~/goesrecv.conf ' responds with "Illegal instruction"

DakotaNelson, 2023-01-06:
I'm having the same problem; only twist on my end is I didn't compile/install goesproc (`-DBUILD_GOESPROC=OFF`).

DakotaNelson, 2023-01-06:
Looks like the same issue as #147

DakotaNelson, 2023-01-06:
Ok, as for helpful details - I'm running on a Pi Zero:

```
pi@goesrcv:~/goestools $ uname -a
Linux goesrcv 5.15.61+ #1579 Fri Aug 26 11:08:59 BST 2022 armv6l GNU/Linux
```

It was installed using this ansible script, with `build_goesrecv` true.

```
- name: Install goestools
  hosts: all
  tasks:

  - name: Install dependencies
    become: yes
    ansible.builtin.package:
      name:
        - zlib1g-dev
        - git
        - librtlsdr-dev
        - cmake
        - gcc
        - g++
      state: present

  - name: Download goestools
    git:
      repo: https://github.com/pietern/goestools.git
      dest: /home/pi/goestools
      recursive: true

  - name: Set up build directory
    file:
      path: /home/pi/goestools/build
      state: directory

  - name: Run cmake for goestools
    command: >
      cmake .. -DCMAKE_INSTALL_PREFIX=/usr/local
      -DBUILD_GOESPROC={{build_goesproc | default(false) | ternary('ON', 'OFF')}}
      -DBUILD_GOESRECV={{build_goesrecv | default(false) | ternary('ON', 'OFF')}}
    args:
      chdir: /home/pi/goestools/build

  - name: Make goestools
    make:
      chdir: /home/pi/goestools/build

  - name: Make install goestools
    become: true
    make:
      chdir: /home/pi/goestools/build
      target: install
```

The config file is:

```
[demodulator]
## Use LRIT mode for GOES-15.
# mode = "lrit"
## Use HRIT mode for GOES-16 or later.
mode = "hrit"
source = "rtlsdr"

# The section below configures the sample source to use.
#
# You can leave them commented out to use the default values for the
# demodulator mode you choose ("lrit" or "hrit"). To use and configure
# any of them, uncomment the section below, and change the demodulator
# source field to match the source you want to use.
#

# [airspy]
# frequency = 1694100000
##
## By default, goesrecv will use the lowest sample rate available.
## This is 2.5 MSPS for the R2 and 3.0 MSPS for the Mini.
## Because different Airspy models support different sample rates,
## it is recommended to leave the "sample_rate" field commented,
## so that it works for either model.
##
# sample_rate = 3000000
# gain = 18
# bias_tee = false

[rtlsdr]
frequency = 1694100000
sample_rate = 2400000
gain = 30
bias_tee = false
device_index = 0

# [nanomsg]
# sample_rate = 2400000
# connect = "tcp://1.2.3.4:5005"
# receive_buffer = 2097152

[costas]
max_deviation = 200e3

[clock_recovery.sample_publisher]
bind = "tcp://0.0.0.0:5002"
send_buffer = 2097152

[quantization.soft_bit_publisher]
bind = "tcp://0.0.0.0:5001"
send_buffer = 1048576

[decoder.packet_publisher]
bind = "tcp://0.0.0.0:5004"
send_buffer = 1048576

# The demodulator stats publisher sends a JSON object that describes
# the state of the demodulator (gain, frequency correction, samples
# per symbol), for every block of samples.
#[demodulator.stats_publisher]
#bind = "tcp://0.0.0.0:6001"

# The decoder stats publisher sends a JSON object for every packet it
# decodes (Viterbi corrections, Reed-Solomon corrections, etc.).
#[decoder.stats_publisher]
#bind = "tcp://0.0.0.0:6002"

# The monitor can log aggregated stats (counters, gauges, and
# histograms) to a statsd daemon. Because this uses UDP, you can keep
# this enabled even if you haven't setup a statsd daemon yet.
#[monitor]
#statsd_address = "udp4://localhost:8125"
```

DakotaNelson, 2023-01-06:
Given [this stackoverflow answer](https://raspberrypi.stackexchange.com/questions/24059/node-js-v0-11-14-exits-with-illegal-instruction), seems likely to be an ARM version problem. I'm gonna see if there are any `cmake` settings I can pass to force the correct version.

```
pi@goesrcv:~/goestools $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 997.08
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 00000000593ea80f
Model           : Raspberry Pi Zero W Rev 1.1
```

DakotaNelson, 2023-01-06:
Wow, ok, apparently newer versions of GCC straight up cannot produce correct ARMv6 executables:

> By default, newer GCC versions do not create correct binaries for ARMv6. Even though you pass the correct -mcpu= flag to gcc, it will create startup code for the newer ARMv7 architecture. Running them on your RasPI Zero will cause an "Illegal Instruction" exception.

https://github.com/Pro/raspi-toolchain

DakotaNelson, 2023-01-06:
Ok, yeah, it looks like the ARM version was correctly detected and GCC just... did it wrong, I think

```
pi@goesrcv:~/goestools $ arm-linux-gnueabihf-readelf -A /usr/local/bin/goesrecv
Attribute Section: aeabi
File Attributes
  Tag_CPU_name: "6"
  Tag_CPU_arch: v6
  Tag_ARM_ISA_use: Yes
  Tag_THUMB_ISA_use: Thumb-1
  Tag_FP_arch: VFPv3
  Tag_Advanced_SIMD_arch: NEONv1
  Tag_ABI_PCS_wchar_t: 4
  Tag_ABI_FP_rounding: Needed
  Tag_ABI_FP_denormal: Needed
  Tag_ABI_FP_exceptions: Needed
  Tag_ABI_FP_number_model: IEEE 754
  Tag_ABI_align_needed: 8-byte
  Tag_ABI_align_preserved: 8-byte, except leaf SP
  Tag_ABI_enum_size: int
  Tag_ABI_VFP_args: VFP registers
  Tag_CPU_unaligned_access: v6
```