pietern/goestools #159: EMWIN date always 12 -31 -1969

nmoon88, 2023-02-17:
Despite the syntax being correct for folder names based on date for EMWIN products the date always remains as 12 -31 -1969 while for everything else the folders have the correct date.

CheaterAdams, 2023-05-14:
I see the same behavior.

RyokoAsakura, 2023-09-21:
I have written 2 scripts that moves the 1969-12-31 in EMWIN to the proper place. The one script is for the current UTC day, and the second script is for whenever your local day is after UTC 00:00. I run them via cron job every 2 minutes during appropriate hours. The only thing you have to do is adjust the hours they get run based on when your local time is UTC 00:00. 

Here's my cron layout for UTC+5 (US Eastern Time)
*/2 0-19/1 * * * bash /home/pi/move1969.sh
*/2 20-23/1 * * * bash /home/pi/nextdaymove1969.sh

And attached are the bash scripts for both. Hope these can help. Don't forget to remove the .txt and also to chmod +x the scripts before you set the cron job for them.

--Ryoko

[move1969.sh.txt](https://github.com/pietern/goestools/files/12678904/move1969.sh.txt)
[nextdaymove1969.sh.txt](https://github.com/pietern/goestools/files/12678905/nextdaymove1969.sh.txt)

nmoon88, 2023-10-08:
I am giving those scripts a try and by the coincidence of me being in the same timezone as the cron layout was designed for I just had to edit the directory of the bash scripts and everything else is as is! All that is left to do is to wait and see what happens for this idea to work around a long standing bug.

nmoon88, 2023-10-08:
Update, the script to move the files to a folder based upon the date in which they were created worked! Now I am just waiting to see what happens with the UTC day script but the fact the first one is working is a really good sign that this one will too.

nmoon88, 2023-10-13:
It only worked once and now I am getting this error from Bash mv: target '/media/nicholas/GOES/goes16_new/emwin/2023-10-13'$'\r''/'$'\r' is not a directory
for some reason all the extra stuff after the date is being added and I can't figure out why. The script looks ok so where is that coming from?

RyokoAsakura, 2024-01-22:
Sorry this reply took so long, life happens I suppose.

I'm unsure where the extra stuff would come from. You could always troubleshoot by breaking out the commands and running them one at a time manually and maybe see if that could be where it's getting stupid...or it could be an encoding thing, so manually transcribing the script into a new file then running that could also help.

stonegrift, 2024-07-20:
I had this issue as well, and I think I may have come up with a work around that easier than moving files (no cron job) By using chrony ntpd and gpsd  I was able to force a regular time update which carried over to the emwin save directory.