pietern/goestools #160: fatal error: proj_api.h: No such file or directory Ubuntu server 22.04 LTS

WakeUpWolfgang, 2023-05-04:
I am trying to build goestools on Ubuntu server 22.04 but I keep getting "fatal error: proj_api.h: No such file or directory"
`matt@sat:~/goestools/build$ sudo make -j2 install
[  1%] Building CXX object src/util/CMakeFiles/util.dir/fs.cc.o
[  1%] Building CXX object src/lib/CMakeFiles/dir.dir/dir.cc.o
[  2%] Building CXX object src/util/CMakeFiles/util.dir/string.cc.o
[  2%] Linking CXX static library libdir.a
[  2%] Built target dir
[  2%] Building CXX object src/lib/CMakeFiles/zip.dir/zip.cc.o
[  2%] Building CXX object src/util/CMakeFiles/util.dir/time.cc.o
[  3%] Linking CXX static library libzip.a
[  3%] Built target zip
[  4%] Building CXX object src/lib/CMakeFiles/timer.dir/timer.cc.o
[  4%] Linking CXX static library libutil.a
[  4%] Built target util
[  4%] Linking CXX static library libtimer.a
[  4%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/ep.c.o
[  4%] Built target timer
[  5%] Building CXX object src/lib/CMakeFiles/unzip.dir/unzip.cc.o
[  5%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/global.c.o
[  7%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/pipe.c.o
[  7%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/poll.c.o
[  7%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/sock.c.o
[  7%] Linking CXX executable unzip
[  7%] Built target unzip
[  7%] Building CXX object src/lib/CMakeFiles/version.dir/version.cc.o
[  8%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/sockbase.c.o
[  8%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/core/symbol.c.o
[  8%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/ctx.c.o
[  9%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/fsm.c.o
[  9%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/pool.c.o
[ 10%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/timer.c.o
[ 11%] Linking CXX static library libversion.a
[ 11%] Built target version
[ 13%] Building CXX object src/lrit/CMakeFiles/lrit.dir/lrit.cc.o
[ 13%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/timerset.c.o
[ 13%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/usock.c.o
[ 14%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/aio/worker.c.o
[ 14%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/alloc.c.o
[ 14%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/atomic.c.o
[ 15%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/chunk.c.o
[ 15%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/chunkref.c.o
[ 15%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/clock.c.o
[ 16%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/closefd.c.o
[ 16%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/efd.c.o
[ 17%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/err.c.o
[ 17%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/hash.c.o
[ 17%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/list.c.o
[ 19%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/msg.c.o
[ 19%] Building CXX object src/lrit/CMakeFiles/lrit.dir/file.cc.o
[ 19%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/condvar.c.o
[ 19%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/mutex.c.o
[ 20%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/once.c.o
[ 20%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/queue.c.o
[ 20%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/random.c.o
[ 21%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/sem.c.o
[ 21%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/sleep.c.o
[ 22%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/strcasecmp.c.o
[ 22%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/strcasestr.c.o
[ 22%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/strncasecmp.c.o
[ 23%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/thread.c.o
[ 23%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/utils/wire.c.o
[ 23%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/devices/device.c.o
[ 25%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/utils/dist.c.o
[ 25%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/utils/excl.c.o
[ 25%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/utils/fq.c.o
[ 26%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/utils/lb.c.o
[ 26%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/utils/priolist.c.o
[ 27%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/bus/bus.c.o
[ 27%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/bus/xbus.c.o
[ 27%] Building C object vendor/nanomsg/src/CMakeFiles/nanomsg.dir/protocols/pipeline/push.c.o
[... 184 lines cut ...]
[ 89%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/options.cc.o
[ 90%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/decoder.cc.o
[ 90%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/demodulator.cc.o
[ 90%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/monitor.cc.o
[ 92%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/datagram_socket.cc.o
[ 92%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/filename.cc.o
[ 92%] Building CXX object src/goesrecv/CMakeFiles/goesrecv.dir/source.cc.o
[ 92%] Linking CXX executable goesrecv
[ 92%] Built target goesrecv
[ 92%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/file_writer.cc.o
[ 92%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/goesproc.cc.o
[ 94%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/gradient.cc.o
[ 94%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/handler_emwin.cc.o
[ 94%] Building CXX object src/goesproc/CMakeFiles/goesproc.dir/handler_goesn.cc.o
In file included from /home/matt/goestools/src/goesproc/map_drawer.h:5,
                 from /home/matt/goestools/src/goesproc/handler_goesn.cc:12:
/home/matt/goestools/src/goesproc/proj.h:16:10: fatal error: proj_api.h: No such file or directory
   16 | #include <proj_api.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/goesproc/CMakeFiles/goesproc.dir/build.make:174: src/goesproc/CMakeFiles/goesproc.dir/handler_goesn.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:2398: src/goesproc/CMakeFiles/goesproc.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
`

kccloy, 2023-05-08:
Noted in proj.org web site under development the proj_api.h no longer exists in the newest iteration of proj.

https://proj.org/development/index.html

mechmyday, 2023-10-08:
See this comment as a workaround https://github.com/pietern/goestools/pull/148#issuecomment-1593582659

pietern, 2024-02-10:
This was fixed in #148 and #167. Thanks for reporting.

closed 2024-02-10 by pietern