pietern/goestools #165: Receiving Images from GOES-16 while pointed at GOES-18

kizmau, 2023-08-15:
I am trying to better understand the receiving capabilities of goestools.  I am pointed at GOES-18 and receiving images, but I am only getting one channel from GOES-16.  Is this due to a configuration error in goesproc-goesr.conf?

I am currently receiving the following images from GOES-18: 
m1 - ch2, ch7, ch13, fc
m2 - ch2, ch7, ch13, fc
fd - ch2, ch7, ch8, ch9, ch13, ch14, ch 15, fc

I am currently receiving the following images from GOES-16:
fd - ch13

My configuration file is as follows:

```
# Store all original GOES-16 products.
[[handler]]
type = "image"
origin = "goes16"
directory = "./goes16/{region:short|lower}/{channel:short|lower}/{time:%Y-%m-%d}"
filename = "GOES16_{region:short}_{channel:short}_{time:%Y%m%dT%H%M%SZ}"
format = "jpg"
json = false

  [[handler.map]]
  path = "/usr/share/goestools/ne/ne_50m_admin_0_countries_lakes.json"

  [[handler.map]]
  path = "/usr/share/goestools/ne/ne_50m_admin_1_states_provinces_lakes.json"

# Store all original GOES-17 products.
[[handler]]
type = "image"
origin = "goes17"
directory = "./goes17/{region:short|lower}/{channel:short|lower}/{time:%Y-%m-%d}"
filename = "GOES17_{region:short}_{channel:short}_{time:%Y%m%dT%H%M%SZ}"
format = "jpg"
json = false

  [[handler.map]]
  path = "/usr/share/goestools/ne/ne_50m_admin_0_countries_lakes.json"

  [[handler.map]]
  path = "/usr/share/goestools/ne/ne_50m_admin_1_states_provinces_lakes.json"

# Added Ability to Decode GOES-18 Products
#
# Store all original GOES-18 products.
[[handler]]
type = "image"
origin = "goes18"
directory = "./goes18/{region:short|lower}/{channel:short|lower}/{time:%Y-%m-%d}"
filename = "GOES18_{region:short}_{channel:short}_{time:%Y%m%dT%H%M%SZ}"
format = "jpg"
json = false

  [[handler.map]]
  path = "/usr/share/goestools/ne/ne_50m_admin_0_countries_lakes.json"

  [[handler.map]]
  path = "/usr/share/goestools/ne/ne_50m_admin_1_states_provinces_lakes.json"

# GOES-18 ABI false color USRADIOGUY CUSTOMLUT NO MAP and jpg.
[[handler]]
type = "image"
origin = "goes18"
regions = [ "fd" ]
channels = [ "ch02", "ch13" ]
directory = "./goes18/{region:short|lower}/CUSTOMLUT/{time:%Y-%m-%d}"
filename = "GOES18_{region:short}_FC_CUSTOMLUT_{time:%Y%m%dT%H%M%SZ}"
format = "jpg"
json = false
[... 417 lines cut ...]
  [[handler.map]]
  path = "/usr/share/goestools/ne/ne_50m_admin_0_countries_lakes.json"

  [[handler.map]]
  path = "/usr/share/goestools/ne/ne_50m_admin_1_states_provinces_lakes.json"

# The following handler takes the same crop from the FD and SH
# products to get more frequent imagery of a smaller area on the
# southern hemisphere.
[[handler]]
type = "image"
origin = "goes15"
regions = [ "fd", "sh" ]
crop = [ -1864, 896, -19, 1043 ]
directory = "./goes15/combine-south/{time:%Y-%m-%d}"
filename = "GOES15_{channel:short}_{time:%Y%m%dT%H%M%SZ}_{region:short}"
format = "jpg"
json = false

  [[handler.map]]
  path = "/usr/share/goestools/ne/ne_50m_admin_0_countries_lakes.json"

  [[handler.map]]
  path = "/usr/share/goestools/ne/ne_50m_admin_1_states_provinces_lakes.json"
```

sky-kable, 2023-10-27:
take a look at this link, the GOES 18 downlink only ever gives ch13 of GOES16 every 180 minutes. (you also get himawari8 every 60 minutes and i'm jealous of that!) https://www.noaasis.noaa.gov/GOES/HRIT/products.html

pietern, 2024-02-10:
Indeed, this is expected. The configuration includes all products for all satellites such that you can use the same configuration for either GOES-16 or GOES-18. It doesn't mean that you get everything when pointing at one of them.

closed 2024-02-10 by pietern