pietern/goestools #174: High Vit after RPI rebuild.

pumatrax, 2024-08-15:
I had some issues with my RPI running goestools and goes-vitality recently. power supply died and memory card was corrupt. After replacing both and rebuilding the pi using the latest 32bit build I am seeing a high vit in goesrecv. It's in the 1000's when normally It used to be 300-400.  

I checked, dongle, lna, cable to feed and I am using a beige grid dish and I think it's all working at expected. When I use satdump I can receive data and files are being created. 

I tried rebuilding goestools several times by can't understand why performance is so poor now. Any thoughts?

LlamaBeing, 2024-08-18:
> I had some issues with my RPI running goestools and goes-vitality recently. power supply died and memory card was corrupt. After replacing both and rebuilding the pi using the latest 32bit build I [...]

I am having a similar problem to yours.  At some point my RPi3b rebooted, upon restarting goesrecv the downlink returns vitavg over 2000 and loss packets exceeding 50.  I have changed the RTL-SDR / SAWBird (GOES) and checked power.  I am at a loss as to what has happened.

sky-kable, 2024-11-22:
Same issue as LlamaBeing, haven't received anything since 10/04. I've tried repeaking the dish, changing SDR, LNA. all to no avail. any suggestions where to go from here would be amazing